serrj-sv/lumi.gateway.mgl03 #1: Syntax Error

DragoshuB, 2020-11-20:
Hello,

I tried to run the new macro but I get syntax error.

Also with the old one when I load the firmware I get wrong firmware file header

Thanks

serrj-sv, 2020-11-20:
> Hello,
> 
> I tried to run the new macro but I get syntax error.
> 
> Also with the old one when I load the firmware I get wrong firmware file header
> 
> Thanks

hello, please show screenshots, I'll take a look on it.  thanks!

serrj-sv, 2020-11-20:
also, you said "old" firmware, what file it is?

DragoshuB, 2020-11-20:
[image]

with old i was referring to the previous mgl03_uart_recovery.ttl file which i have downloaded this morning.

Regarding the firmware I tried both 

mgl03_1.4.7_0065_mod20201110.uart
mgl03_1.4.7_0065_mod20201116.uart

And both of them triggered the same error.

serrj-sv, 2020-11-20:
> [image]
> 
> with old i was referring to the previous mgl03_uart_recovery.ttl file which i have downloaded this morning.
> 
> Regarding the firmware I tried both
> 
> mgl03_1.4.7_0065_mod20201110.uart
> mgl03_1.4.7_0065_mod20201116.uart
> 
> And both of them triggered the same error.

looks like you downloaded HTML file instead of script itself
you should click on "raw" button on github page, after you click on file you want to download.
or use this direct link:
https://github.com/serrj-sv/lumi.gateway.mgl03/raw/main/uart_recovery/mgl03_uart_recovery.ttl

DragoshuB, 2020-11-20:
I cannot believe it. Indeed this was the issue. Now the files are transferred. Thank you for you fast support.

serrj-sv, 2020-11-20:
> I cannot believe it. Indeed this was the issue. Now the files are transferred. Thank you for your fast support.

glad to help!
have fun!

DragoshuB, 2020-11-20:
👍

closed 2020-11-20 by DragoshuB