serrj-sv/lumi.gateway.mgl03 #3: Question: Can it be flashed on 1.4.7_0065

sameersbn, 2020-11-23:
Hi! I upgraded my device to the latest version and want to integrate the device with home assistant.
Can I use you instructions from https://github.com/serrj-sv/lumi.gateway.mgl03/tree/main/uart_recovery to flash the firmware?

serrj-sv, 2020-11-23:
> Hi! I upgraded my device to the latest version and want to integrate the device with home assistant.
> Can I use you instructions from https://github.com/serrj-sv/lumi.gateway.mgl03/tree/main/uart_recovery to flash the firmware?

yes, using this instruction you can flash modified firmware, which works with Home Assistant

closed 2020-11-24 by serrj-sv

sameersbn, 2020-11-25:
thanks for the quick reply @serrj-sv .
I messed up and damaged the rx and tx tracks on the board while making the solder connections.
Any idea if there are other points on the board that will give me the rx and tx terminals?

serrj-sv, 2020-11-25:
> thanks for the quick reply @serrj-sv .
> I messed up and damaged the rx and tx tracks on the board while making the solder connections.
> Any idea if there are other points on the board that will give me the rx and tx terminals?

here you go:
https://raw.githubusercontent.com/serrj-sv/lumi.gateway.mgl03/main/media/mgl03_back_uart_eth.jpg

sameersbn, 2020-11-25:
thanks a lot. successfully installed the modified firmware. thanks for the effort (sent you a donation too :slightly_smiling_face: ).

serrj-sv, 2020-11-25:
Got it! Thanks for contrubution, have fun with your gateway :)