serrj-sv/lumi.gateway.mgl03 #8: Bluetooth devices data

TDenisM, 2020-12-10:
Hi,
where can I get data of connected bluetooth devices? On the old original firmware it was possible to get it with this command
`killall silabs_ncp_bt; silabs_ncp_bt /dev/ttyS1 1
`
On new custom firmware I get nothing from ble devices.

AlexxIT, 2020-12-10:
Do you use my component?

TDenisM, 2020-12-10:
No - I use Nodered for data gathering. Where can I find data inside gateway?

TDenisM, 2020-12-12:
Ok, I installed HA, HACS, your component for Gateway 3 - as a result I can get info from 1 device of 4, moreover - I can see only one parameter (Conductivity) of 3 (Conductivity, humidity, ambient brightness), more moreover - the value of the parameter doesn't update - it differ from the actual value in Xiaomi app. I don't know where the problem is - in gateway firmware or in component.

plutosherry, 2021-01-03:
me too. After I upgraded 1.4.7_0115, the same problem occurred. The data of the Bluetooth device could not be obtained and updated, even after adding it again.

plutosherry, 2021-01-03:
By the way, my previous use of mgl03_1.4.7_0065_mod20201211 has been normal.