serrj-sv/lumi.gateway.mgl03 #9: Scrip not working, os recognize uart.

broda1125, 2020-12-10:
Hello there.
Before I've reared all forums I've updated my gateway 3 to the newest firmware... finally I've found your recovery solution, bought myself uart adapter, connected to Win 10 and everything was running fine till I've left my pc unattended for couple minutes. When I got back the transfer popup dissappeard and since that moment I can't connect any uart. Gateway works fine with devices and Mi Home but I wanted to use it with Home Assistant. When using Windows, system see the uart adapter as COM 2 but while using macro nothing happens when plug in power adapter. While using Linus exactly the same story. I've tried also backup uart pinout without result.
Thanks in advance for any tips!

framiso, 2020-12-15:
Hello @broda1125  I have more or less the same issue as you,. In my case I think that is caused by 5v supply. 

I bougth a UART CP2104  and I have 2 pins, one 3.3v and another of 5v but in the pics you don't need to connect on any of them...  But i check the voltage between TX and TR and when the gateway is connected to micro USB power shows 3v but when I turn of the gateway and connect Tera program shows 5v.

Someone know hoy to set the USB into 3.3v? Or maybe I need other USB adapter? 

Any ideas @serrj-sv?

Thanks

broda1125, 2020-12-16:
Hello @framiso as far as I know when you are facing this issue you have to use power stabilizer to reduce voltage from 5v to 3.3v. I guess better option will be to buy other UART adapter as I just did. Sadly I have accidentally burned my gateway and now I'm awaiting delivery ;) Good UART should be CP2102 which I will test around upcoming weekend (I have also FTDI232 uart adapter which theoretically should work as well).

closed 2020-12-28 by broda1125