serrj-sv/lumi.gateway.mgl03 #11: Flashing via telnet guide

iNiSiRe, 2020-12-19:
Can I flash that firmware via telnet by dd or something or only via UART? I have 1.6.5 so I have access to telnet but firmware has limitation such as mqtt on 127.0.0.1, don't want every time fix this.

zvldz, 2020-12-19:
https://gist.github.com/zvldz/b40b4873e3c4c1a64ac536e8ce5dbdad

This tutorial will be here on github soon.

iNiSiRe, 2020-12-19:
Wow, I didn't expect answer so quick. (: Thank you so match. You guys made great job, never flashing by UART be so easy. Plastic case disassembling be more complex step than flashing. How can I donate?

iNiSiRe, 2020-12-19:
By the way, may be you can give some advises with another lumi device, I'll be very appreciate. Can I ask there or how should I do?

zvldz, 2020-12-19:
>  How can I donate?

https://buymeacoff.ee/serrj
author of github

> Can I ask there or how should I do?
you can try to ask in our group https://t.me/xiaomi_gw_v3_hack

closed 2020-12-19 by iNiSiRe