serrj-sv/lumi.gateway.mgl03 #16: Which UART should I order ?

Tatoo974, 2021-01-14:
Hi , sorry I’m not used with UART usb stick. Could someone help me to choose on AliExpress a good one for soldering my gateway 3 ? Thanks a lot

Yiivgeny, 2021-01-14:
Works for me https://aliexpress.ru/item/32673467746.html
[image]

Tatoo974, 2021-01-14:
Ok thanks a lot !
And I have just to solder the TX,RX and GND to the board at the righ places ? How to choose 3.3v mode ?
Should I have to buy an external power supply ?
Thank you

Yiivgeny, 2021-01-14:
I connect TX->RX, RX->TX, GND->GND, gateway connected to USB port on my computer (with inlcuded cable), uart stick to another port, and it's all.

Tatoo974, 2021-01-15:
Thanks a lot !

closed 2021-01-15 by Tatoo974