serrj-sv/lumi.gateway.mgl03 #18: Flash gateway using non solder option

aserno, 2021-02-01:
I was trying to downgrade the gateway with this method. Got the pcb clip connected on the Board with the ttl to usb 3.3v jumper set(as was suggested) windows sees the usb module after a driver install(com4)

next step connect it with de tera term.
Script says connect it with poweroff. Then power on.. after that.. nothing. What should i expect to see after the gateway boots up? Script doesn’t change after the “power on” what do i miss?

sinad07, 2021-02-02:
Неправильное подключение ,поменяйте местами rx - tx

-- Отправлено из мобильного приложения Яндекс.Почты

marrubio, 2021-02-06:
I have the same issue with soldering option. 

The usb to serial converter seems to work correctly once the drivers are installed, but there seems to be no communication, I have checked the welds with a tester and they seem to work without apparent problems, current firmware version v1.4.6_0043

PinkLolicorn, 2021-02-10:
I just solved my problem similar to this one. I have tried many times with different UART sticks with no success - always no output when trying to connect. 

Then I stepped at the post regarding serial connections on the sparkfun containing this line:
> So the RX from one device should go to the TX of the other, and vice-versa.

So at the end I figured out that in fact you have to connect Rx on gateway to Tx on UART, not Rx to Rx. It may be not that obvious to someone who is making serial connection for the first time.

serrj-sv, 2021-02-10:
its basically mentioned manual:

2. connect UART:
UART Tx <--> TP4 (Gateway Rx)
UART Rx <--> TP11 (Gateway Tx)
UART GND <--> TP8 (Gateway GND)

PinkLolicorn, 2021-02-10:
> its basically mentioned manual: 2. connect UART: UART Tx <--> TP4 (Gateway Rx) UART Rx <--> TP11 (Gateway Tx) UART =

Yes, I just noticed it and wanted to add a note in my first post. I probably assumed my Tx-Tx and Rx-Rx makes sense and proceeded to the next step. I think it is possible to mistake to be made and is a probable cause of the issues.

marrubio, 2021-02-12:
Thank you very much. That was de problem, the "TX" and "RX" cables were NOT crossed, I didn't read that part of the tutorial

Tualua, 2024-11-27:
Hello!
[image]

My gateway throws garbage to UART. Is this a problem with the gateway itself or am I doing something wrong?
I am using FTDI TTL to USB cable and solderless clip
[image]

bitrate is 115200, but I tried to use other speed with the same result