serrj-sv/lumi.gateway.mgl03 #21: Macro no asking about FW file

Trapasin, 2021-02-06:
Hello!
When I load macro, I step-by-step go to load rtkboot_11520
after it is loading macro dont asking about FW file
only if I make Disconnect in Tera macro will ask about file.
After select file - progress bar is empty and I have message that its fail in few minutes

I try to change port speed in Tera and Windows, trying different rtkboot_***
no ways
I try to wait 2 hours after loading rtkboot_115200 - no result. Macro dont asking about FW file

Trapasin, 2021-02-06:
I use this downloaded files:

[issue.zip](https://github.com/serrj-sv/lumi.gateway.mgl03/files/5937571/issue.zip)

PinkLolicorn, 2021-02-10:
I'm getting same results, I analyzed the macro code and confirmed in the Tera Term's Control > Show Macro Window with the result: "mgl03_uart_recovery:67:wait '<RealTek>'" where the hang is.

The issue occurs at the same logic in the Linux script. At first, I though maybe only Linux version wasn't working, but it appears that both are affected.

I tried unplugging UART and assembling my hub, I'm getting solid yellow/orange light. No connection in app after a while. I tried adding it again by holding button 10 seconds a couple of times without reaction, still solid light.

Can you confirm the orange light, or it is not the case here? I think I somehow bricked my hub.

PinkLolicorn, 2021-02-10:
I tried running Linux script again and there is a line saying "dram_com_addr_rot test fails." which is a little bit concerning. Anyways, sharing full log with everyone:

`# ./mgl03_uart_recovery.expect 
Xiaomi Gateway 3 flashing script
Requires:
  sx (package lrzsz)
  stty (package coreutils)

Connect Gateway to UART, disconnect it from power

Available bootloaders:
[0] rtkboot_460800.bin
[1] rtkboot_57600.bin
[2] rtkboot_38400.bin
[3] rtkboot_230400.bin
[4] rtkboot_115200.bin
Choose bootloader (0): 4
Using bootloader from: rtkboot_115200.bin
Selected bootloader speed: 115200

Available firmwares:
[0] mgl03_1.4.6_0012_mod20210127.uart
[1] mgl03_1.4.7_0115_mod20201222.uart
Choose firmware (0): 1
Using firmware from: mgl03_1.4.7_0115_mod20201222.uart

kernel size: 20e804
rootfs size: bca002
rootfs offset: a0c0e83a

Enter serial device name (/dev/ttyUSB0): 
Connecting to /dev/ttyUSB0

spawn [open ...]

Power ON Gateway NOW!
uart ok
strap pin:0x412b8ae2
enable spi-nand
ROM ver:v1.1, sig:455cc27, time:2016.01.04-18:42+0800, CPU(400 MHz), DDR2(533 MHz)
Enter ROM console
<RealTek>dbgmsg 3
dbg level: 0x00000003
<RealTek>ri 0 1 1
load efuse ok
rom_progress: 0x0200006d
0xb8000780: 0x00c78e0f
0xb8000784: 0x15408c2f
0xb8000788: 0x0000bfe0
0xb800078c: 0x00000000
0xb8000790: 0x00000000
0xb8000794: 0x00000000
0xb8000798: 0x00000000
0xb800079c: 0x00000000
0xb80007a0: 0x00000000
0xb80007a4: 0x00000000
0xb80007a8: 0x00000000
0xb80007ac: 0x00000000
0xb80007b0: 0x00000000
0xb80007b4: 0x00000000
0xb80007b8: 0x00000000
0xb80007bc: 0x00000000
0xb80007c0: 0x00000000
0xb80007c4: 0x00000000
0xb80007c8: 0x00000000
0xb80007cc: 0x00000000
0xb80007d0: 0x00000000
0xb80007d4: 0x00000000
0xb80007d8: 0x00000000
0xb80007dc: 0x00000000
0xb80007e0: 0x00000000
0xb80007e4: 0x00000000
0xb80007e8: 0x00000000
0xb80007ec: 0x00000000
0xb80007f0: 0x00000102
0xb80007f4: 0x50c23826
0xb80007f8: 0x00000005
0xb80007fc: 0x00000001
load_efuse_data(361): loc:0x00000200, offset:0x00000040, word_enb:0x0000000e
loc:0x00000200, ctrl:0x000000c7
loc:0x00000200, ctrl2:0x00000000
0x200	c7 00 00 00 00 00 00 00 
load_efuse_data(361): loc:0x00000208, offset:0x00000041, word_enb:0x0000000c
loc:0x00000208, exec_from_sram_addr:0xbfe01540
0x208	40 15 e0 bf 00 00 00 00 
load_efuse_data(353): read eFuse done 0x00b bytes 
ctrl:			0x000000c7
ctrl2:			0x00000000
exec_from_sram_addr:	0xbfe01540
boot_addr:		0x00000000
img_offset:		0x00000000
init_dram_val:		0x00000000
heap_start_addr:		0x00000000
heap_len:		0x00000000
erase_start_addr:		0x00000000
rom_progress: 0x0400006d

dram_init.c : ss_init_rxi310()

dram_init.c : ms_ctrl_0_map=0xb8142000
ss_init_rxi310(1902):bond:0x0000000a

 dram_init_clk_frequency ,ddr_freq=1066 (Mbps), 533 (MHZ) 
mck_ck = 0, 1084

JSW :  dram_init_rxi310 ,dev_map=0xb8142000 

JSW :  dram_init_rxi310 ,dram_info=0x9fe03dd8 

JSW :  dram_init_rxi310 001

dfi_rate=2,dram_period=3752

JSW :  dram_init_rxi310 002

SDR D1  dram_info-> dev-> device_type=2 ,cr_twr=3 

DDR2  dram_info-> dev-> device_type=2

DDR2  dram_info-> mode_reg-> bst_len=0

SDR   dq_width=2

SDR   page=3

SDR  dram_info-> dev-> bank =1

cr_bst_len =0

SDR   dfi_rate=2

SDR  dq_width=2

ddr_type =2

SDR   cas_rd /(dfi_rate)=3

SDR  cas_wr /(dfi_rate) =3

SDR  dev_map-> dcr =0x00000222

SDR  dev_map-> iocr=0x00204000

dev_map-> emr1=0x00000006

dev_map-> mr=0x00000672

SDR  dev_map-> drr=0x0912be1c

SDR  dev_map-> tpr0=0x000066c4

SDR  dev_map-> tpr1=0x002a9104

SDR  dev_map-> tpr2=0x00000042

DRAM init disable

DRAM init enable

DRAM init is done , jump to DRAM

SDR  init done , dev_map=0xb8142000
Enter dram_auto_size_detect_rxi310, page_size = 3, bank_size = 1 

DDR2  dram_info-> mode_reg-> bst_len=0

Detect page_size = 2KB (3)

Detect bank_size =	4 banks(0x00000001)

Detect dram size = 64MB (0x04000000)

 dram_init.c : dram_calibration_turn_on_odt.

Perform sample dram testing
=======start dram_addr_rot test=======
rotate 0 0xa0400100 passed
rotate 4 0xa0400100 passed
rotate 8 0xa0400100 passed
rotate 12 0xa0400100 passed
rotate 16 0xa0400100 passed
rotate 20 0xa0400100 passed
rotate 24 0xa0400100 passed
rotate 28 0xa0400100 passed
dram_addr_rot test completed.
=======start dram_com_addr_rot test=======
~rotate 0 0xa0400100 passed
~rotate 4 0xa0400100 passed
~rotate 8 0xa0400100 passed
~rotate 12 0xa0400100 passed
decr addr(0xa04000fc): 0xff07ffff != pattern(0xff03ffff) dram_com_addr_rot, 794

dram_com_addr_rot test fails.

init ddr fail
rom_progress: 0x0900006d
<RealTek>
spawn sh -c sx -vv -X -b rtkboot_115200.bin < /dev/ttyUSB0 > /dev/ttyUSB0
Sending rtkboot_115200.bin, 773 blocks: Give your local XMODEM receive command now.
Bytes Sent:  98944   BPS:3528                            

Transfer complete
ecv data ok len: 0x00018280
<RealTek>
<RealTek>j 80000000
Jump to 0x80000000
`

serrj-sv, 2021-02-10:
your device is dead :(

PinkLolicorn, 2021-02-10:
> your device is dead :(

Do you think it is possible to revive it, or it was killed for good? Before I have found this repository I remember plugging in AC adapter and 3.3V from UART at the same time. Chip on the USB stick was overheating, so I unplugged it. Never checked if it works before starting the processed described here, so it was possible the hub was damaged already.

serrj-sv, 2021-02-11:
I don't think it is possible to revive it, the RAM in this device is embedded in main SoC, so you have to replace the main chip. Easier just to buy new device.

> > your device is dead :(
> 
> Do you think it is possible to revive it, or it was killed for good? Before I found this repository I remember plugging in AC adapter and 3.3V from UART at the same time. Chip on the USB stick was overheating, so I unplugged it. Never checked if it works before starting the processed described here, so it is possible the hub was damaged already.

Trapasin, 2021-02-11:
problem here: I download wrong bootloader
try to download correct file

*__________________________*

*Gorchevsky Vladimir*

*mob.: +**996 555 787222 (WhatsUp, Viber, Telegram) *

*Skype: vladimir_gorchevsky*

*www.linkedin.com/in/vladimirgor* <http://www.linkedin.com/in/vladimirgor>

closed 2021-02-11 by Trapasin