serrj-sv/lumi.gateway.mgl03 #24: ZNDMWG02LM (евро версия, без штекера) v1.4.6_0043

TRibyn, 2021-02-11:
Добрый день!!!
Я новичок в этом деле. По незнанию купил Gateway 3 v1.4.6_0043 - ZNDMWG02LM (Telnet c паролем) . Хочу подключить выключатели Lonsonho. Как я понял - нужно понизить прошивку... С пайкой дружу. Интересует очередность действий: HassWP установил, UART имеется. Не совсем понимаю про telnet. Или для понижения прошивки не требуется распароливать telnet?
Заранее спасибо!!!

PinkLolicorn, 2021-02-11:
Здравствуй,

После выполнения uart_recovery (https://github.com/serrj-sv/lumi.gateway.mgl03/blob/main/uart_recovery/README.md) вы сможете использовать устройство с XiaomiGateway3, который предварительно установлен с HassWP. См. 

Больше на: https://github.com/AlexxIT/XiaomiGateway3/wiki#main-method

TRibyn, 2021-02-13:
Спасибо за помощь. Все подключил, все работает))))

DimonAver, 2021-04-23:
Прошил XiaomiGateway3 согласно инструкции, но выключатель Lonsonho не спаривается. Как Вам удалось? Поделитесь инструкцией плиз.