serrj-sv/lumi.gateway.mgl03 #26: mgl03_uart_recovery.ttl script stops after "j 80000000" command

dwojtkowiak, 2021-02-14:
For ZNDMWG02LM, macro stops after Jump to command.

serrj-sv, 2021-02-15:
Hi, please check troubleshooting area. Make sure you downloaded bootloader properly

> For ZNDMWG02LM, macro stops after Jump to command.

dwojtkowiak, 2021-02-15:
Thanks, but it is something else (soldering is fine, tx/rx OK, I get all prompts etc). I have downloaded script, reviewed it, tried executing commands without script. 

1. Script executes OK up to loading bootloader file
2. Once bootloader upload completed, it stops
3. I modified scripts a bit by removing "clear screen" and noticed it stops after  j 80000000
4. I got message "Unknow command" for "xmod a0a0000"

Do you know what might be an issue please? Thanks

serrj-sv, 2021-02-15:
post here full output pls

dwojtkowiak, 2021-02-15:
==MGL03 UART Recovery v231120_01==

   >>Power ON Gateway NOW!<<

uart ok
strap pin:0x412b8ae2
enable spi-nand
ROM ver:v1.1, sig:455cc27, time:2016.01.04-18:42+0800, CPU(400 MHz), DDR2(533 MHz)
Enter ROM console
<RealTek>
<RealTek>dbgmsg 3
dbg level: 0x00000003
<RealTek>ri 0 1 1
load efuse ok
rom_progress: 0x0200006d
0xb8000780: 0x00c78e0f
0xb8000784: 0x15408c2f
0xb8000788: 0x0000bfe0
0xb800078c: 0x00000000
0xb8000790: 0x00000000
0xb8000794: 0x00000000
0xb8000798: 0x00000000
0xb800079c: 0x00000000
0xb80007a0: 0x00000000
0xb80007a4: 0x00000000
0xb80007a8: 0x00000000
0xb80007ac: 0x00000000
0xb80007b0: 0x00000000
0xb80007b4: 0x00000000
0xb80007b8: 0x00000000
0xb80007bc: 0x00000000
0xb80007c0: 0x00000000
0xb80007c4: 0x00000000
0xb80007c8: 0x00000000
0xb80007cc: 0x00000000
0xb80007d0: 0x00000000
0xb80007d4: 0x00000000
0xb80007d8: 0x00000000
0xb80007dc: 0x00000000
0xb80007e0: 0x00000000
0xb80007e4: 0x00000000
0xb80007e8: 0x00000000
0xb80007ec: 0x00000000
0xb80007f0: 0x00000102
0xb80007f4: 0x658b1044
0xb80007f8: 0x00000005
0xb80007fc: 0x00000001
load_efuse_data(361): loc:0x00000200, offset:0x00000040, word_enb:0x0000000e
loc:0x00000200, ctrl:0x000000c7
loc:0x00000200, ctrl2:0x00000000
0x200   c7 00 00 00 00 00 00 00
load_efuse_data(361): loc:0x00000208, offset:0x00000041, word_enb:0x0000000c
loc:0x00000208, exec_from_sram_addr:0xbfe01540
0x208   40 15 e0 bf 00 00 00 00
load_efuse_data(353): read eFuse done 0x00b bytes
ctrl:                   0x000000c7
ctrl2:                  0x00000000
exec_from_sram_addr:    0xbfe01540
boot_addr:              0x00000000
img_offset:             0x00000000
init_dram_val:          0x00000000
heap_start_addr:                0x00000000
heap_len:               0x00000000
erase_start_addr:               0x00000000
rom_progress: 0x0400006d

dram_init.c : ss_init_rxi310()

dram_init.c : ms_ctrl_0_map=0xb8142000
ss_init_rxi310(1902):bond:0x0000000a

 dram_init_clk_frequency ,ddr_freq=1066 (Mbps), 533 (MHZ)
mck_ck = 0, 1084

JSW :  dram_init_rxi310 ,dev_map=0xb8142000

JSW :  dram_init_rxi310 ,dram_info=0x9fe03dd8

JSW :  dram_init_rxi310 001

dfi_rate=2,dram_period=3752

JSW :  dram_init_rxi310 002

SDR D1  dram_info-> dev-> device_type=2 ,cr_twr=3

DDR2  dram_info-> dev-> device_type=2

DDR2  dram_info-> mode_reg-> bst_len=0

SDR   dq_width=2

SDR   page=3

SDR  dram_info-> dev-> bank =1

cr_bst_len =0

SDR   dfi_rate=2

SDR  dq_width=2

ddr_type =2

SDR   cas_rd /(dfi_rate)=3

SDR  cas_wr /(dfi_rate) =3

SDR  dev_map-> dcr =0x00000222

SDR  dev_map-> iocr=0x00204000

dev_map-> emr1=0x00000006

dev_map-> mr=0x00000672

SDR  dev_map-> drr=0x0912be1c

SDR  dev_map-> tpr0=0x000066c4

SDR  dev_map-> tpr1=0x002a9104

SDR  dev_map-> tpr2=0x00000042

DRAM init disable

DRAM init enable

DRAM init is done , jump to DRAM

SDR  init done , dev_map=0xb8142000
Enter dram_auto_size_detect_rxi310, page_size = 3, bank_size = 1

DDR2  dram_info-> mode_reg-> bst_len=0

Detect page_size = 2KB (3)

Detect bank_size =      4 banks(0x00000001)

Detect dram size = 64MB (0x04000000)

 dram_init.c : dram_calibration_turn_on_odt.

Perform sample dram testing
=======start dram_addr_rot test=======
rotate 0 0xa0400100 passed
rotate 4 0xa0400100 passed
rotate 8 0xa0400100 passed
rotate 12 0xa0400100 passed
rotate 16 0xa0400100 passed
rotate 20 0xa0400100 passed
rotate 24 0xa0400100 passed
rotate 28 0xa0400100 passed
dram_addr_rot test completed.
=======start dram_com_addr_rot test=======
~rotate 0 0xa0400100 passed
~rotate 4 0xa0400100 passed
~rotate 8 0xa0400100 passed
~rotate 12 0xa0400100 passed
decr addr(0xa04000f4): 0xff0fffff != pattern(0xff0bffff) dram_com_addr_rot, 794

dram_com_addr_rot test fails.

init ddr fail
rom_progress: 0x0900006d
<RealTek>
<RealTek>xmrx 80000000
recv data ok len: 0x00018280
<RealTek>j 80000000
Jump to 0x80000000

serrj-sv, 2021-02-15:
what happens when you switch on gateway without script? does it boot
properly?

dwojtkowiak, 2021-02-15:
Loads and I get RealTek> prompt.

serrj-sv, 2021-02-15:
can you post output for this as well

dwojtkowiak, 2021-02-15:
I will get this output tomorrow and share here (not around laptop anymore). Thanks for help with this

dwojtkowiak, 2021-02-16:
<RealTek>
<RealTek>
<RealTek>
<RealTek>uart ok
strap pin:0x412b8ae2
enable spi-nand
ROM ver:v1.1, sig:455cc27, time:2016.01.04-18:42+0800, CPU(400 MHz), DDR2(533 MHz)
load efuse ok
init IP ok
rom_progress: 0x0600006d
load_data_from_storage(260): 0xbfe01540, 0x00000000, 0xbfd16f44
load_data_from_spi_nand_flash(70): 0xbfe01540, 0x00000000, 0xbfe03e18
 <ecc decode fail>
ECC_DECODE_FAIL,ecc_start_addr = 0xbfe01540
check_image_header(72): h(69,72,00,20), s(69,72,61,6d)
img sig err
rom_progress: 0x0d00006d
load_data_from_storage(260): 0xbfe01540, 0x00000080, 0xbfd16f44
load_data_from_spi_nand_flash(70): 0xbfe01540, 0x00000080, 0xbfe03e18
ECC_DECODE_FAIL,ecc_start_addr = 0xbfe01540
check_image_header(72): h(ff,ff,ff,ff), s(69,72,61,6d)
img sig err
rom_progress: 0x0d00006d
load_data_from_storage(260): 0xbfe01540, 0x00000100, 0xbfd16f44
load_data_from_spi_nand_flash(70): 0xbfe01540, 0x00000100, 0xbfe03e18
ECC_DECODE_FAIL,ecc_start_addr = 0xbfe01540
check_image_header(72): h(ff,ff,ff,ff), s(69,72,61,6d)
img sig err
rom_progress: 0x0d00006d
load_data_from_storage(260): 0xbfe01540, 0x00000180, 0xbfd16f44
load_data_from_spi_nand_flash(70): 0xbfe01540, 0x00000180, 0xbfe03e18
check_image_header(72): h(ff,ff,ff,ff), s(69,72,61,6d)
img sig err
rom_progress: 0x0d00006d
load img fail(0xffffffff)
rom_progress: 0x1100006d
load code fail (0xffffffff)
rom_progress: 0x0b00006d

dram_init.c : ss_init_rxi310()

dram_init.c : ms_ctrl_0_map=0xb8142000
ss_init_rxi310(1902):bond:0x0000000a

 dram_init_clk_frequency ,ddr_freq=1066 (Mbps), 533 (MHZ)
mck_ck = 0, 1084

JSW :  dram_init_rxi310 ,dev_map=0xb8142000

JSW :  dram_init_rxi310 ,dram_info=0x9fe03f20

JSW :  dram_init_rxi310 001

dfi_rate=2,dram_period=3752

JSW :  dram_init_rxi310 002

SDR D1  dram_info-> dev-> device_type=2 ,cr_twr=3

DDR2  dram_info-> dev-> device_type=2

DDR2  dram_info-> mode_reg-> bst_len=0

SDR   dq_width=2

SDR   page=3

SDR  dram_info-> dev-> bank =1

cr_bst_len =0

SDR   dfi_rate=2

SDR  dq_width=2

ddr_type =2

SDR   cas_rd /(dfi_rate)=3

SDR  cas_wr /(dfi_rate) =3

SDR  dev_map-> dcr =0x00000222

SDR  dev_map-> iocr=0x00204000

dev_map-> emr1=0x00000006

dev_map-> mr=0x00000672

SDR  dev_map-> drr=0x0912be1c

SDR  dev_map-> tpr0=0x000066c4

SDR  dev_map-> tpr1=0x002a9104

SDR  dev_map-> tpr2=0x00000042

DRAM init disable

DRAM init enable

DRAM init is done , jump to DRAM

SDR  init done , dev_map=0xb8142000
Enter dram_auto_size_detect_rxi310, page_size = 3, bank_size = 1

DDR2  dram_info-> mode_reg-> bst_len=0

Detect page_size = 2KB (3)

Detect bank_size =      4 banks(0x00000001)

Detect dram size = 64MB (0x04000000)

 dram_init.c : dram_calibration_turn_on_odt.

Perform sample dram testing
=======start dram_addr_rot test=======
rotate 0 0xa0400100 passed
rotate 4 0xa0400100 passed
rotate 8 0xa0400100 passed
rotate 12 0xa0400100 passed
rotate 16 0xa0400100 passed
rotate 20 0xa0400100 passed
rotate 24 0xa0400100 passed
rotate 28 0xa0400100 passed
dram_addr_rot test completed.
=======start dram_com_addr_rot test=======
~rotate 0 0xa0400100 passed
~rotate 4 0xa0400100 passed
~rotate 8 0xa0400100 passed
~rotate 12 0xa0400100 passed
decr addr(0xa04000f4): 0xff0fffff != pattern(0xff0bffff) dram_com_addr_rot, 794

dram_com_addr_rot test fails.

init ddr fail
rom_progress: 0x0900006d
load_data_from_storage(260): 0xa0500000, 0x00000000, 0xbfd16f4c
load_data_from_spi_nand_flash(70): 0xa0500000, 0x00000000, 0xbfe03e30
 <ecc decode fail>
ECC_DECODE_FAIL,ecc_start_addr = 0xa0500000
check_image_header(72): h(69,72,00,20), s(62,6f,6f,74)
img sig err
rom_progress: 0x0d00006d
load_data_from_storage(260): 0xa0500000, 0x00000080, 0xbfd16f4c
load_data_from_spi_nand_flash(70): 0xa0500000, 0x00000080, 0xbfe03e30
ECC_DECODE_FAIL,ecc_start_addr = 0xa0500000
check_image_header(72): h(ff,ff,ff,ff), s(62,6f,6f,74)
img sig err
rom_progress: 0x0d00006d
load_data_from_storage(260): 0xa0500000, 0x00000100, 0xbfd16f4c
load_data_from_spi_nand_flash(70): 0xa0500000, 0x00000100, 0xbfe03e30
ECC_DECODE_FAIL,ecc_start_addr = 0xa0500000
check_image_header(72): h(ff,ff,ff,ff), s(62,6f,6f,74)
img sig err
rom_progress: 0x0d00006d
load_data_from_storage(260): 0xa0500000, 0x00000180, 0xbfd16f4c
load_data_from_spi_nand_flash(70): 0xa0500000, 0x00000180, 0xbfe03e30
 <ecc decode fail>
ECC_DECODE_FAIL,ecc_start_addr = 0xa0500000
check_image_header(72): h(ff,ff,ff,ff), s(62,6f,6f,74)
img sig err
rom_progress: 0x0d00006d
load img fail(0xffffffff)
rom_progress: 0x1100006d
<RealTek>

serrj-sv, 2021-02-17:
> dram_com_addr_rot test fails.
> 
> init ddr fail

looks like device is dead, see similar case [here](https://github.com/serrj-sv/lumi.gateway.mgl03/issues/21#issuecomment-777011348)