serrj-sv/lumi.gateway.mgl03 #27: pair doesn't work

minoccio, 2021-02-16:
I have successfully flashed with 1.4.6_0012_mod20210127 version, using uart and solder, and gateway works perfectly in Home Assistant, the proble is that I can't pair new child devices in HA, but if I pair same device with Mi Home app, it works in HA

serrj-sv, 2021-02-16:
> I have successfully flashed with 1.4.6_0012_mod20210127 version, using uart and solder, and gateway works perfectly in Home Assistant, the proble is that I can't pair new child devices in HA, but if I pair same device with Mi Home app, it works in HA

Hi, I'm glad that flashing went well! :) For HA integration-related question and issues please visit corresponding project page:
https://github.com/AlexxIT/XiaomiGateway3

meceo, 2021-02-17:
@minoccio congrats that you have done it. Could you help me with the issue of using UART? https://github.com/serrj-sv/lumi.gateway.mgl03/issues/25