serrj-sv/lumi.gateway.mgl03 #37: Unable to get past the bootloader

rafaelpini, 2023-11-10:
I have attempted flashing with various bootloaders, but no matter what approach I take, it gets stuck at "<RealTek>xmrx 80000000".

The dialog box that is supposed to show XMODEM activity never displays any progress. It seems to time out, and consequently, the window closes.

Below is my log:

`==MGL03 UART Recovery v231120_01==

   >>Power ON Gateway NOW!<<

uart ok

strap pin:0x412b8ae2

enable spi-nand

ROM ver:v1.1, sig:455cc27, time:2016.01.04-18:42+0800, CPU(400 MHz), DDR2(533 MHz)

Enter ROM console

<RealTek>

<RealTek>dbgmsg 3

dbg level: 0x00000003

<RealTek>ri 0 1 1

load efuse ok

rom_progress: 0x0200006d

0xb8000780: 0x00c78e0f

0xb8000784: 0x15408c2f

0xb8000788: 0x0000bfe0

0xb800078c: 0x00000000

0xb8000790: 0x00000000

0xb8000794: 0x00000000

0xb8000798: 0x00000000

0xb800079c: 0x00000000

0xb80007a0: 0x00000000

0xb80007a4: 0x00000000

0xb80007a8: 0x00000000

0xb80007ac: 0x00000000

0xb80007b0: 0x00000000

0xb80007b4: 0x00000000

0xb80007b8: 0x00000000

0xb80007bc: 0x00000000

0xb80007c0: 0x00000000

0xb80007c4: 0x00000000

0xb80007c8: 0x00000000

0xb80007cc: 0x00000000

0xb80007d0: 0x00000000

0xb80007d4: 0x00000000

0xb80007d8: 0x00000000

0xb80007dc: 0x00000000

0xb80007e0: 0x00000000

0xb80007e4: 0x00000000

0xb80007e8: 0x00000000

0xb80007ec: 0x00000000

0xb80007f0: 0x00000102

0xb80007f4: 0x2c6f3e13

0xb80007f8: 0x00000005

0xb80007fc: 0x00000001

load_efuse_data(361): loc:0x00000200, offset:0x00000040, word_enb:0x0000000e

loc:0x00000200, ctrl:0x000000c7

loc:0x00000200, ctrl2:0x00000000

0x200	c7 00 00 00 00 00 00 00 

load_efuse_data(361): loc:0x00000208, offset:0x00000041, word_enb:0x0000000c

loc:0x00000208, exec_from_sram_addr:0xbfe01540

0x208	40 15 e0 bf 00 00 00 00 

load_efuse_data(353): read eFuse done 0x00b bytes 

ctrl:			0x000000c7

ctrl2:			0x00000000

exec_from_sram_addr:	0xbfe01540

boot_addr:		0x00000000

img_offset:		0x00000000

init_dram_val:		0x00000000

heap_start_addr:		0x00000000

heap_len:		0x00000000

erase_start_addr:		0x00000000

rom_progress: 0x0400006d

dram_init.c : ss_init_rxi310()

dram_init.c : ms_ctrl_0_map=0xb8142000

ss_init_rxi310(1902):bond:0x0000000a

 dram_init_clk_frequency ,ddr_freq=1066 (Mbps), 533 (MHZ) 

mck_ck = 0, 1084

JSW :  dram_init_rxi310 ,dev_map=0xb8142000 

JSW :  dram_init_rxi310 ,dram_info=0x9fe03dd8 

JSW :  dram_init_rxi310 001

dfi_rate=2,dram_period=3752

JSW :  dram_init_rxi310 002

SDR D1  dram_info-> dev-> device_type=2 ,cr_twr=3 

DDR2  dram_info-> dev-> device_type=2

DDR2  dram_info-> mode_reg-> bst_len=0

SDR   dq_width=2

SDR   page=3

SDR  dram_info-> dev-> bank =1

cr_bst_len =0

SDR   dfi_rate=2

SDR  dq_width=2

ddr_type =2

SDR   cas_rd /(dfi_rate)=3

SDR  cas_wr /(dfi_rate) =3

SDR  dev_map-> dcr =0x00000222

SDR  dev_map-> iocr=0x00204000

dev_map-> emr1=0x00000006

dev_map-> mr=0x00000672

SDR  dev_map-> drr=0x0912be1c

SDR  dev_map-> tpr0=0x000066c4

SDR  dev_map-> tpr1=0x002a9104

SDR  dev_map-> tpr2=0x00000042

DRAM init disable

DRAM init enable

DRAM init is done , jump to DRAM

SDR  init done , dev_map=0xb8142000

Enter dram_auto_size_detect_rxi310, page_size = 3, bank_size = 1 

DDR2  dram_info-> mode_reg-> bst_len=0

Detect page_size = 2KB (3)

Detect bank_size =	4 banks(0x00000001)

Detect dram size = 64MB (0x04000000)

 dram_init.c : dram_calibration_turn_on_odt.

Perform sample dram testing

=======start dram_addr_rot test=======

rotate 0 0xa0400100 passed

rotate 4 0xa0400100 passed

rotate 8 0xa0400100 passed

rotate 12 0xa0400100 passed

rotate 16 0xa0400100 passed

rotate 20 0xa0400100 passed

rotate 24 0xa0400100 passed

rotate 28 0xa0400100 passed

dram_addr_rot test completed.

=======start dram_com_addr_rot test=======

~rotate 0 0xa0400100 passed

~rotate 4 0xa0400100 passed

~rotate 8 0xa0400100 passed

~rotate 12 0xa0400100 passed

~rotate 16 0xa0400100 passed

~rotate 20 0xa0400100 passed

~rotate 24 0xa0400100 passed

~rotate 28 0xa0400100 passed

dram_com_addr_rot test completed.

JSW 20141208:=======start dram_byte_access test=======

JSW 20141208:pattern[0](0x01234567) 0xa0400100 pass

JSW 20141208:pattern[1](0xfedcba98) 0xa0400100 pass

JSW 20141208:pattern[2](0xa5a5a5a5) 0xa0400100 pass

JSW 20141208:pattern[3](0x5a5a5a5a) 0xa0400100 pass

JSW 20141208:pattern[4](0x5a5aa5a5) 0xa0400100 pass

JSW 20141208:pattern[5](0xa5a55a5a) 0xa0400100 pass

dram_byte_access test completed.

=======start dram_half_word_access test=======

pattern[0](0x01234567) 0xa0400100 pass

pattern[1](0xfedcba98) 0xa0400100 pass

pattern[2](0xa5a5a5a5) 0xa0400100 pass

pattern[3](0x5a5a5a5a) 0xa0400100 pass

pattern[4](0x5a5aa5a5) 0xa0400100 pass

pattern[5](0xa5a55a5a) 0xa0400100 pass

dram_half_word_access test completed.

=======start dram_walking_of_1 test=======

pattern[0](0x00000001) 0xa0400100 passed

pattern[1](0x00000002) 0xa0400100 passed

pattern[2](0x00000004) 0xa0400100 passed

pattern[3](0x00000008) 0xa0400100 passed

pattern[4](0x00000010) 0xa0400100 passed

pattern[5](0x00000020) 0xa0400100 passed

pattern[6](0x00000040) 0xa0400100 passed

pattern[7](0x00000080) 0xa0400100 passed

pattern[8](0x00000100) 0xa0400100 passed

pattern[9](0x00000200) 0xa0400100 passed

pattern[10](0x00000400) 0xa0400100 passed

pattern[11](0x00000800) 0xa0400100 passed

pattern[12](0x00001000) 0xa0400100 passed

pattern[13](0x00002000) 0xa0400100 passed

pattern[14](0x00004000) 0xa0400100 passed

pattern[15](0x00008000) 0xa0400100 passed

pattern[16](0x00010000) 0xa0400100 passed

pattern[17](0x00020000) 0xa0400100 passed

pattern[18](0x00040000) 0xa0400100 passed

pattern[19](0x00080000) 0xa0400100 passed

pattern[20](0x00100000) 0xa0400100 passed

pattern[21](0x00200000) 0xa0400100 passed

pattern[22](0x00400000) 0xa0400100 passed

pattern[23](0x00800000) 0xa0400100 passed

pattern[24](0x01000000) 0xa0400100 passed

pattern[25](0x02000000) 0xa0400100 passed

pattern[26](0x04000000) 0xa0400100 passed

pattern[27](0x08000000) 0xa0400100 passed

pattern[28](0x10000000) 0xa0400100 passed

pattern[29](0x20000000) 0xa0400100 passed

pattern[30](0x40000000) 0xa0400100 passed

pattern[31](0x80000000) 0xa0400100 passed

dram_walking_of_1 test completed.

=======start dram_walking_of_0 test=======

pattern[0](0xfffffffe) 0xa0400100 passed

pattern[1](0xfffffffd) 0xa0400100 passed

pattern[2](0xfffffffb) 0xa0400100 passed

pattern[3](0xfffffff7) 0xa0400100 passed

pattern[4](0xffffffef) 0xa0400100 passed

pattern[5](0xffffffdf) 0xa0400100 passed

pattern[6](0xffffffbf) 0xa0400100 passed

pattern[7](0xffffff7f) 0xa0400100 passed

pattern[8](0xfffffeff) 0xa0400100 passed

pattern[9](0xfffffdff) 0xa0400100 passed

pattern[10](0xfffffbff) 0xa0400100 passed

pattern[11](0xfffff7ff) 0xa0400100 passed

pattern[12](0xffffefff) 0xa0400100 passed

pattern[13](0xffffdfff) 0xa0400100 passed

pattern[14](0xffffbfff) 0xa0400100 passed

pattern[15](0xffff7fff) 0xa0400100 passed

pattern[16](0xfffeffff) 0xa0400100 passed

pattern[17](0xfffdffff) 0xa0400100 passed

pattern[18](0xfffbffff) 0xa0400100 passed

pattern[19](0xfff7ffff) 0xa0400100 passed

pattern[20](0xffefffff) 0xa0400100 passed

pattern[21](0xffdfffff) 0xa0400100 passed

pattern[22](0xffbfffff) 0xa0400100 passed

pattern[23](0xff7fffff) 0xa0400100 passed

pattern[24](0xfeffffff) 0xa0400100 passed

pattern[25](0xfdffffff) 0xa0400100 passed

pattern[26](0xfbffffff) 0xa0400100 passed

pattern[27](0xf7ffffff) 0xa0400100 passed

pattern[28](0xefffffff) 0xa0400100 passed

pattern[29](0xdfffffff) 0xa0400100 passed

pattern[30](0xbfffffff) 0xa0400100 passed

pattern[31](0x7fffffff) 0xa0400100 passed

dram_walking_of_0 test completed.

=======start memcpy_test test=======

memcpy 0 0xa0400100 passed

 

memcpy 4 0xa0400100 passed

 

memcpy 8 0xa0400100 passed

 

memcpy 12 0xa0400100 passed

 

memcpy 16 0xa0400100 passed

 

memcpy 20 0xa0400100 passed

 

memcpy 24 0xa0400100 passed

 

memcpy 28 0xa0400100 passed

 memcpy_test test completed.

  init ddr ok

rom_progress: 0x0800006d

<RealTek>

<RealTek>xmrx 80000000
`

Another peculiar issue I've noticed is that the bootloader binary file I've been using is getting rewritten, and its size drops to 0 KB.

I´ve already tried inverting the TX/RX cable, using another USB port, another USB power adaptor and nothing works.

Can someone plese tell me what can I do?

e6a6, 2023-11-14:
There seems to be a bug in the latest version of Tera Term (currently 2023.10.15 (Ver 5.0)). I had the exact same problem as you, but there were no problems using the previous version, Tera Term 4.106.

rafaelpini, 2023-11-14:
> There seems to be a bug in the latest version of Tera Term (currently 2023.10.15 (Ver 5.0)). I had the exact same problem as you, but there were no problems with the previous version, Tera Term 4.106.

You're absolutely right; this was the cause of my problem as well. Thanks for the insight! I switched to the final release of version 4, which is 4.107, and everything worked as expected.

@serrj-sv, it might be beneficial to include a link to version 4.107 in the Readme, as it seems to be stable. Here is the link to the Tera Term repository pointing to that version: 
https://github.com/TeraTermProject/teraterm/releases/tag/v4.107

closed 2023-11-14 by rafaelpini